bigtreetech/KNOMI #2: Macros on PDF do not format Correctly and getting `Macro G28 called recursively` homing errors from provided macros

BoothyBoothy, 2023-05-07:
Hi BTT
1. Copying the PDF macro content directly from the PDF does not work because all formatting content is lost and Klipper fails to boot due to parsing errors. The gcode lines need indenting to stop the parsing issue.
2. After including the Knomi Macros Print_Start fails with a repeated homing error stating `Macro G28 called recursively` (3 times). 

Please advise and thanks Boothy

hdragoon, 2023-05-08:
You may did something wrong, as I did the same as instruction and no problem at all.

BoothyBoothy, 2023-05-08:
Thanks for the response. Hmmm, are you using a Klicky Probe or Tap? May be the external Klicky variable causing the issues? I doubt if my install will last long anyway, the PFC is less for it, great fun but not at the cost of print quality IMHO.

BoothyBoothy, 2023-05-08:
Further to my last comment. I use a Klicky Probe on my V2. If  I exclude the Klicky Variables.cfg the BTT homing macro works without errors and I get a House symbol on the Knomi display whilst homing, not tried anything else yet. The BTT macro therefore seems to work but does not allow for any external homing Macro reference like Klicky external Macros. I will need to consider what is the most tidy way forward with this and the Klicky. Thanks Boothy

dirkgo, 2023-05-09:
Same here. Also with Klicky Probe. Will figure out later today, how to fix it.

Surion79, 2023-05-17:
@BoothyBoothy I created a Pull Request to fix the G28 issue by using a dedicated `_KNOMI_STATUS` macro

Cyondil, 2023-05-18:
> @BoothyBoothy I created a Pull Request to fix the G28 issue by using a dedicated `_KNOMI_STATUS` macro

I have the Knomi and tested your PR and it doesn't seem to work at all.  If I use the lines above pointing to BED_MESH_CALIBRATE, the Knomi accurately shows levelling.  When I substitute in _KNOMI_STATUS, it doesn't work.  So seems like the logic is looking for the combination for the variable of the probing variable change and the BED_MESH_CALIBRATE combination.

Cyondil, 2023-05-18:
Had a whole thing typed up with a fix for homing this, but realized that while testing, my individual axis homing broke, so I am not confident enough to offer it as a fix.

Bazeone, 2023-06-06:
The changes worked for me.

sgermain06, 2023-07-02:
Yeah, I took a similar approach and added a `CUSTOM_HOMING` macro that the code is listening for, but it doesn't feel right. There has to be a better way to do this.

PeterGarthAmos, 2023-07-13:
Hi, all, I'm not that clued up with the code, please could you explain what you did exactly if you don't mind. (dumb it down) I have the same issue and have not been able to resolve it. I have klicky. the knomi macro causes Macro G28 called recursively.

3DMat, 2023-07-19:
> Same here. Also with Klicky Probe. Will figure out later today, how to fix it.
> 
> Bed Mesh works with this macro now:
> 
> [image]
> 
> Still no solution for Homing G28 without being recursively called.
> 
> Solved it now. Changed the source code to request a created macro variable "homing" in "_Probe_variables" instead of G28. Set it True at the beginning of "homing_override" and False at the end of macro "Park_Toolhead".

hello sorry can you put a screen of the macro for homing if it's possible I'm trying the thousand recommended methods including using the _knomi_status updating the knomi with vcore and the bed mesh doesn't even work (only works using the one posted here)
A thousand thanks

silentstorm21, 2023-10-01:
i'm having this issue now

BoothyBoothy, 2023-10-01:
I found this to work well on my Minion Bed Slinger with Klicky.
https://github.com/bigtreetech/KNOMI/pull/5
The only thing that does not appear to work is the print finished macro? I have no problems with anything else.
Best Boothy

nephilim667, 2023-10-26:
Hello, i have integrated a knomi on my Creality K1. As i have issues to modify Knomi firmware, i have found solution to make it works with sandard firmware.
You can found details of macros modifications here :
https://github.com/D3vil-Design/K1-Hardware-Mods/issues/4#issue-1962019105