bigtreetech/KNOMI #9: Mounting plate missing from STL.

grimlokason, 2023-05-21:
Hello,

The mounting plate is not anymore in the STL here : 

https://github.com/bigtreetech/KNOMI/blob/master/Customized%20Printed%20Parts/Mounting%20Plate%20%26%20StealthBurner%20Main%20Body%20for%20KNOMI.STL

zcohenld, 2023-05-22:
They just updated it with the mounting plate, but be aware, they uploaded the two parts as one file and most importantly the mounting plate is not flat on the build plate. So break it into two parts and use the place on face tool in your slicer to lay the plate flat.

thisiskeithb, 2023-05-23:
Here's a zip with the KNOMI StealthBurner & separated mounting plate properly aligned: [Mounting Plate & StealthBurner Main Body for KNOMI.STL.zip](https://github.com/bigtreetech/KNOMI/files/11543902/Mounting.Plate.StealthBurner.Main.Body.for.KNOMI.STL.zip)

chromusphil, 2023-05-31:
The new version was rushed and the plate isn't sitting flat, please provide separate STLs and or FIX the plate so its laying flat on the bed.

grimlokason, 2023-06-04:
@bigtreetech is it possible to accept the pull request ? 

Thank you.

closed 2023-06-06 by grimlokason