bigtreetech/KNOMI #14: Printing percentages not correct

Nelus82, 2023-06-06:
Mainsail and klipperscreen gives a lower percentage of the print than KNOMI.

chromusphil, 2023-08-05:
I’ve experienced this however I did notice that the % matched that displayed in the 12864 screen on my v2.4 so I’m assuming the completeness used by Mainsail/Klipperscreen is calculated differently to the Knomi/12864

chromusphil, 2023-08-06:
[image]
Further to my reply above

Mcrcheli, 2023-08-31:
Seems to calculate the percentage from the time you press print. As you get into the print the percentage converge and I have seen the numbers match in the 75-80 range.

nephilim667, 2023-10-26:
Hello, i hvae found something relative to this.
Knomi use the Display_Status Api function of Klipper.
As you can see, Display status use value from M73 if GCode containt it :
[image]

As i have the same issue, i have tried to disable 'Support remaining time' is prusaslicer to avoid emmiting M73 in GCode.
Seem to work. Now Knomi show the same value as i have on my K1 Screen and on Fluidd