bigtreetech/KNOMI #16: Is there a way for Knomi to show temp by Tool?

TheSin-, 2023-07-02:
I'd like to add a knot to each of my tools and show the temp of the tool it's attached to, is there a way to accomplish this easily?  To T0 will show the temp for T0 and T1 will show the temp for T1 etc etc

M4dMike, 2023-08-10:
Sounds like a cool idea :)

This would require costume code on each Knomi, but can definitely be achieved.

Have a look at line 1007 to 1015 in [main.cpp](https://github.com/bigtreetech/KNOMI/blob/master/Firmware/src/main.cpp)

    String nameStr3 = doc["temperature"]["tool0"]["actual"].as<String>();
    String nameStr4 = doc["temperature"]["tool0"]["target"].as<String>();
    String nameStr5 = doc["state"]["flags"]["printing"].as<String>();
    String nameStr6 = doc["state"]["flags"]["paused"].as<String>();

    bedtemp_actual = (uint16_t)((doc["temperature"]["bed"]["actual"].as<double>())*100);
    bedtemp_target = (uint16_t)((doc["temperature"]["bed"]["target"].as<double>())*100);
    tooltemp_actual = (uint16_t)((doc["temperature"]["tool0"]["actual"].as<double>())*100);
    tooltemp_target = (uint16_t)((doc["temperature"]["tool0"]["target"].as<double>())*100);`

As you can see, its referencing temps for `tool0` -> change that too `tool1` and build + flash it to your second KNOMI

TheSin-, 2023-08-10:
excellent!  I'll order up an other one to see if I can figure it out, this will be great if I can get it running!

TheSin-, 2023-08-10:
Do you know of an example to make it a configuration option on startup, like the screen used for the wifi setup?  That way they can all have one firmware that I can just select the tool id on?

Trying to figure out the codebase for configuration options.

M4dMike, 2023-08-10:
Exactly, adding a 4th input field and storing this information makes it dynamic and would probably help to get this change back into mainline. Awesome!

TheSin-, 2023-08-11:
Into the testing phase now!  Thanks again for leading me into the right direction, I'm going to make a PR once I know 100% it's working and saving.

[image]

M4dMike, 2023-08-11:
Awesome 😀
Happy to help, knomi has a lot of unused potential 😉

mcbeenb, 2023-11-20:
Awesome work guys!!