bigtreetech/KNOMI #18: 2 KNOMI's will not power on by USB-C connection to computer or show up in device list

sprior, 2023-08-01:
I've got 2 KNOMI's which show no signs of power when connected via USB-C to either of two Win 11 laptops (and don't know up as ports in device manager, but do power up when connected via USB-C wall wart.  The problem is that I'm looking to customize their firmware and see no way to do that.  I have other ESP32 devices with power up and flash just fine from my laptop.

I've otherwise got a VS Code dev environment set up just fine for ESP32 devices on this laptop so I'm pretty stumped why the KNOMI's aren't working.  Any ideas?

M4dMike, 2023-08-08:
Had the same issue, USB-A to USB-C works for me, Knomi is showing up in device manager as "USB-SERIAL CH340 (COM3)"

Did you manage to get to a working state? I am struggling with flashing, as uploading the firmware fails... 

> A fatal error occurred: Failed to connect to ESP32: No serial data received.

Malygros, 2023-08-10:
> I was using a USB-C to USB-C cable to plug them into my computer. By any chance is a USB-A (computer) to USB-C (KNOMI) cable required to make it work?

Did you mange do get it work? Got the same error. Even with USB-A to USB-C. 
USB-SERIAL CH340 (COM4) is recognized in the device Manager.

sprior, 2023-08-10:
It's been working for me with a USB-C to USB-A cable, though the requirement to use one isn't documented anywhere that I've seen.  I also have to hold the boot button before plugging it in, then it's recognized by the flash utility in PlatformIO.

Fragmon, 2023-08-10:
Is this possible that knomi make a Mistake with tracing? They add a 10k on 3.3V not gnd
[image]
[image]

M4dMike, 2023-08-10:
Nope, it's correct and it works fine after all.
If you want to set it into boot mode, disconnect it from power, press the button an plug it in while the button is pressed

sprior, 2023-08-10:
@M4dMike Flash mode only works with a USB-A to USB-C cable for me when connecting to my laptop.  If I use a USB-C to USB-C cable the KNOMI doesn't get power at all when in boot mode or not.  That same USB-C to USB-C cable works with a USB-C power supply.

M4dMike, 2023-08-10:
Same for me.

Fragmon, 2023-08-10:
> @Fragmon Nope, it's correct and it works fine after all. If you want to set it into boot mode, disconnect it from power, press the button an plug it in while the button is pressed

I know this:) but it dont change the Mode. It's always in serial.

sprior, 2023-08-10:
@M4dMike I am not at all an expert on this, but I think I've heard that USB-C devices use resistors to signal their need for power or ability to provide it.  So I thought @Fragmon's comment about resistors not being correct might be related to power delivery and not the ability to enter boot mode.  It could also explain why a USB-A to USB-C cable would work and a USB-C to USB-C does not.

Ybalrid, 2023-09-10:
all I know is that the modem control signals from the serial are not able to reset the device from the USB port. You need to press the boot button to upload new firmware, and once new firmware is uploaded, you need to disconnect and reconnect the device for it to startup again.