bigtreetech/KNOMI #19: Failed to connect to ESP32: No serial data received.

ThomasDennisSp, 2023-08-07:
I tryed the methode that the other person had and others tryed but they had no effect i cant get vs to flash firmware to my knomi it gets detected and everything by vs but the flashing will not work

M4dMike, 2023-08-08:
Get the exact same error.

>Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\esp32dev\firmware.bin
esptool.py v4.5.1
Serial port COM3
Connecting......................................
> A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

COM3 is detected as USB-Serial CH340

M4dMike, 2023-08-08:
After reading the documentation on the ESP32 i got it working:

1. Build image
2. Press Boot button while plugging the KNOMI into the USB Port
3. Upload Image

ThomasDennisSp, 2023-08-08:
ok the comment above helped me in figuring it out it is important that your knomi is not getting pover over the jst then
1. build image
2. press and keep holding the boot button for the complete time it takes 
3. connect the usb-c to your pc (still holding the button)
4. upload image (still holding button till success massage comes up)
5. let the boot button go unplug usb c
6. connect it to your printer
7. profit

closed 2023-08-08 by ThomasDennisSp