bigtreetech/KNOMI #25: KAMP bed mesh level now disabled

Mcrcheli, 2023-09-07:
After finally getting the knomi to work as intended, my last issue is that the code disables KAMP. KAMP renames bed_mesh_calibrate to _bed_mesh_calibrate. Wondering if knomi  renaming it to bed_mesh_calibrate_base is the issue. The Kamp.cfg is read only so not sure what to look at. Thanks!

Ybalrid, 2023-09-09:
Having the same issue right now! I may just set the gcode variables on my own around my leveling stuff in PRINT_START instead see if it can trigger the animation without breaking KAMP

Mcrcheli, 2023-09-09:
I commented out the mesh leveling lines of code. This allows Kamp to wrk but the icon doesn’t work. I have an email into BTT and found my original issue but they are working on the Kamp issue. When I get the reply I will post it here.

Ybalrid, 2023-09-09:
I have done a hack in the mean time, sadly it requires KNOMI firmware modification 

I now set the `variable_probing` inside `PRINT_START` around my bed mesh (and quad gantry) leveling calls. But for KNOMI to fetch the state, you need to replace what gcode_macro is looked at in the API response from Moonraker in a few places in `main.cpp`. Not ideal

Ybalrid, 2023-09-10:
@Mcrcheli if you do not mind a hacky solution: This is the patch to the KNOMI firmware to make it look for that "probing" gcode variable into PRINT_START https://github.com/Ybalrid/KNOMI-UNIT-01/commit/9f0720a12bec801c4792ece9430cd3b0d4f2c912

This is how it is used:
```ini
[gcode_macro PRINT_START] #   Use PRINT_START for the slicer starting script - please customise for your slicer of choice
variable_probing:False
gcode:
    # ...
    SET_GCODE_VARIABLE MACRO=PRINT_START VARIABLE=probing VALUE=True
    # Do your meshing and your leveling and all that stuff
    SET_GCODE_VARIABLE MACRO=PRINT_START VARIABLE=probing VALUE=False
```

nephilim667, 2023-10-26:
Hello, i have integrated a knomi on my Creality K1. As i have issues to modify Knomi firmware, i have found solution to make it works with sandard firmware.
You can found details of macros modifications here : 
https://github.com/D3vil-Design/K1-Hardware-Mods/issues/4#issue-1962019105

Ybalrid, 2024-01-18:
@Sidkrsk in your slicer, you may want to add something like this to your layer change custom code 

```gcode
;[layer_z]
M73 P{(layer_num*100)/total_layer_count}
```

This will insert a GCODE command `M73 PX` where X is the percentage completion of the model, computed from the height (you may be able to do this with the time estimate, but that sounds more complicated to do)

The above is using **Prusa Slicer** templating syntax. For other slicers, you may need to do something different.

Sidkrsk, 2024-01-19:
I still managed to add it to KAMP. I used the latest firmware from December 14th.

printer.cfg

```
[gcode_macro _KNOMI_STATUS]
variable_homing: False
variable_probing: False
variable_qgling: False
variable_heating_nozzle: False
variable_heating_bed: False
gcode:

[gcode_macro M109]
rename_existing: M109.1
gcode:
  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=heating_nozzle VALUE=True
  M109.1 {rawparams}
  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=heating_nozzle VALUE=False

[gcode_macro M190]
rename_existing: M190.1
gcode:
  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=heating_bed VALUE=True
  M190.1 {rawparams}
  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=heating_bed VALUE=False

[gcode_macro G28]
rename_existing: G28.1
gcode:
  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=True
  G28.1 {rawparams}
  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=False

#[gcode_macro BED_MESH_CALIBRATE]
#rename_existing: BTT_BED_MESH_CALIBRATE
#gcode:
#  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=True
#  BTT_BED_MESH_CALIBRATE
#  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=False

# For printer with QGL
[gcode_macro QUAD_GANTRY_LEVEL]
rename_existing: BTT_QUAD_GANTRY_LEVEL
gcode:
  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=qgling VALUE=True
  BTT_QUAD_GANTRY_LEVEL
  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=qgling VALUE=False
```

Adaptive_Mesh.cfg

```
[gcode_macro BED_MESH_CALIBRATE]
rename_existing: _BED_MESH_CALIBRATE
gcode:

    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}                                # Gather all object points
    {% set bed_mesh_min = printer.configfile.settings.bed_mesh.mesh_min %}                                                          # Get bed mesh min from printer.cfg
    {% set bed_mesh_max = printer.configfile.settings.bed_mesh.mesh_max %}                                                          # Get bed mesh max from printer.cfg
    {% set probe_count = printer.configfile.settings.bed_mesh.probe_count %}                                                        # Get probe count from printer.cfg
    {% set kamp_settings = printer["gcode_macro _KAMP_Settings"] %}                                                                 # Pull variables from _KAMP_Settings
    {% set verbose_enable = kamp_settings.verbose_enable | abs %}                                                                   # Pull verbose setting from _KAMP_Settings
    {% set probe_dock_enable = kamp_settings.probe_dock_enable | abs %}                                                             # Pull probe dockable probe settings from _KAMP_Settings
    {% set attach_macro = kamp_settings.attach_macro | string %}                                                                    # Pull attach probe command from _KAMP_Settings
    {% set detach_macro = kamp_settings.detach_macro | string %}                                                                    # Pull detach probe command from _KAMP_Settings
    {% set mesh_margin = kamp_settings.mesh_margin | float %}                                                                       # Pull mesh margin setting from _KAMP_Settings
    {% set fuzz_amount = kamp_settings.fuzz_amount | float %}                                                                       # Pull fuzz amount setting from _KAMP_Settings
    {% set probe_count = probe_count if probe_count|length > 1 else probe_count * 2  %}                                             # If probe count is only a single number, convert it to 2. E.g. probe_count:7 = 7,7
    {% set max_probe_point_distance_x = ( bed_mesh_max[0] - bed_mesh_min[0] ) / (probe_count[0] - 1)  %}                            # Determine max probe point distance
[... 75 lines cut ...]
            )) }

            {action_respond_info("KAMP adjustments successful. Happy KAMPing!")}

        {% else %}

            {action_respond_info("No objects detected! Check your gcode and make sure that EXCLUDE_OBJECT_DEFINE is happening before BED_MESH_CALIBRATE is called. Defaulting to regular meshing.")}
            G4 P5000                                                                                                                # Wait 5 seconds to make error more visible
        {% endif %}

    {% endif %}

    {% if probe_dock_enable == True %}
        {attach_macro}                                                                                                              # Attach/deploy a probe if the probe is stored somewhere outside of the print area
    {% endif %}
    
    SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=True
    _BED_MESH_CALIBRATE mesh_min={adapted_x_min},{adapted_y_min} mesh_max={adapted_x_max},{adapted_y_max} ALGORITHM={algorithm} PROBE_COUNT={points_x},{points_y}
    SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=False

    {% if probe_dock_enable == True %}
        {detach_macro}                                                                                                              # Detach/stow a probe if the probe is stored somewhere outside of the print area
    {% endif %}                                                                                                                     # End of verbose

```

FxWoody, 2025-02-25:
To use KAMP, simply change KNOMI.cfg like so .... works on my Qidi Plus 4
`[gcode_macro _BED_MESH_CALIBRATE]
rename_existing: BTT_BED_MESH_CALIBRATE
gcode:
  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=True
  BTT_BED_MESH_CALIBRATE {rawparams}
  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=False`