bigtreetech/KNOMI #31: KNOMI GIF

CBoismenu, 2023-10-14:
Not a problem with KNOMI itself but a problem with the GIF file subblied by BigTreeTech.

The GIF for leveling in the ZIP file is not the same as the one used in the original firmware.

The one in the firmware is a hotend going up an down, the onr in the ZIP file is like the circles that would be created by throwing a rock in water.

I've searched and could not find the original leveling GIF used by BTT.

Would it be possible to get a copy of the original leveling GIF.

Thanks,

lvlacleod, 2023-10-14:
[image]

CBoismenu, 2023-10-14:
Thank's that was quick.

closed 2023-10-17 by CBoismenu