bigtreetech/KNOMI #32: knomi no wifi website?

dannler, 2023-10-18:
sup guys i have the KNOMI already flashed, ready for wifi connection... my issue is when i connect my pc or phone to the knomi wifi network it then opens up a msn page, it doesnt take me to the knomi wifi page to link it to my network? anyone had this kind of issue?

lvlacleod, 2023-10-19:
I had a problem with mine and wifi and it all came down to my PC was using 5GHz and knomi was using 2.4GHz. I had to force my PC to connect using 2.4 only.

dannler, 2023-10-19:
> I had a problem with mine and wifi and it all came down to my PC was using 5GHz and knomi was using 2.4GHz. I had to force my PC to connect using 2.4 only.

thank you, that sorted one issue, now i am stuck with KNOMI has lost touch with the printer.

closed 2023-10-19 by dannler