bigtreetech/KNOMI #37: Knomi V2: Doesn't retain WiFi information.

MKloberg, 2023-11-12:
When multiple networks are present, Knomi V2 onboarding works and connects to the specified network after 1st restart.
After 1st power cycle, it goes back to onboarding screen and we have to enter the WiFi password again. This repeats itself.
Each time this is done, a new network entry is added to the WiFi networks table.

If this is the wrong repository, please indicate where the correct one is for Knomi V2.

Rowan2233, 2023-11-15:
Hey bro, i am sorry to bring this experience for you. To make sure of this problem, we have this two questions, please help:
1. Did KNOMI 2 suddenly back to the "Hello" page when you repower ? Or it have the “LOGO+circle loading”page for a while and then it back to the "Hello” page？
2. Do you installed the antenna on KNOMI 2 during you  repower KNOMI 2?

MKloberg, 2023-11-15:
Hello, the antenna was installed the whole time. After power is applied, the Logo and circle page comes up, circles several times and then it goes into the Hello page. Someone else that I'm working with is seeing the exact same thing.

MKloberg, 2023-11-18:
Now that the firmware is available I was able to monitor the serial output of V2 and understand this a little better.
The normal onboarding looks like the monitor below, all good.
After the power is removed and plugged back in, it doesn't find that network again and goes back into setup mode.

A theory I have is that it finds multiple networks of the same name in my case, because I'm on a Google mesh network.
The multiple AP's I have, appear all with the same name in the list at the top of the setup screen.

[image]

Perhaps after reboot, it's connecting to the wrong access point (too far away), fails the check and pops back into setup mode.

Unfortunately, I can't simulate a cold boot up, because the serial port isn't available in vscode, do you know a way around that? ( I tried the button for +5secs but suspect that resets the EEprom too)

The other thing that's been funky is that sometimes while going through these steps, the Knomi suddenly emits a SSID that is all garbled up, like something was not translated correctly somewhere for the SSID. 
I did that recently when I tried to manually set the STA SSID and password to see what that would do. 

Log:

 *  Executing task: C:\Users\mklob\.platformio\penv\Scripts\platformio.exe device monitor --environment knomiv2 

--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
btn: 100
btn: 100
btn: 100
btn: 800
EEPROM reset factory
x�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x420d71ae
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4

------------- Knomi startup -----------

SPI Flash: 
  Size: 16777216
  Speed: 80000000
  Mode: 0
SPI PSRAM: 
  Found: 1
  Size: 8386279

------------------------------------------

******** LIS2DW12 startup *****

LIS2DW12 found!

******** LIS2DW12 init ok *****

knomi_config from default & INIT EEPROM
mdns hostname: KNOMI
knomi_config_require: AP
ap ssid: BTT-KNOMI
ap pwd:
ap ip: 192.168.20.1
ap mac: 4A:27:E2:F0:DB:38

dnsServer: 1
knomi_config_require: STA
mDNS start ok!
Scaning...
Scaning...
Scaning...
Scaning...
lv_wifi_status: Scaning...0

Scaning...
Scaning...
Scaning...
Scaning...
Scaning...
Scaning...
Scaning...
Scaning...
Scaning...
Scan ok!
knomi_config write to EEPROM
knomi_config_require: 16
para: 16
klipper ip: 192.168.86.29
knomi_config write to EEPROM
knomi_config_require: 10
para: 10
knomi_config_require: STA

sta ssid: MySpectrumWiFi10-5G
sta pwd: xxxxxxxxxxxxxxxx
.lv_wifi_status: 1
......................................................................sta ip: 192.168.86.49
lv_wifi_status: 2

MKloberg, 2023-11-21:
Can we please get some traction on this 'show stopper' issue? 
Several customers are reporting the same thing now, rendering the device useless on their networks.
Please give this issue some level of escalation.
Thank you. :-)

Foose83, 2023-11-29:
Hello. I entered a shorter ssid and it works correctly now.

bigtreetech, 2023-12-06:
Hi Please try the latest version of firmware, this should solve the bug of not saving WiFi when password or name are long.
KNOMI2:
https://github.com/bigtreetech/KNOMI/blob/master/KNOMI2/Firmware/knomi2_firmware.bin
KNOMI1:
https://github.com/bigtreetech/KNOMI/blob/master/KNOMI1/Firmware/knomi1_firmware.bin
Follow the steps below to update:
https://github.com/bigtreetech/KNOMI/#firmware-update

closed 2023-12-06 by bigtreetech

alcaron, 2023-12-24:
Yeah the "choose file" button on Mac doesn't do anything...