bigtreetech/KNOMI #38: Knomi v2 Wi-Fi

3dbrooks, 2023-11-12:
Confirmed, Wi-Fi password is not retained after power is removed

3dbrooks, 2023-11-12:
Knomi V2...

3dbrooks, 2023-11-15:
I believe it briefly shows the logo and then back to the hello page with antenna connected . When I reconnect to it, the klipper IP is still in there but I must select my network from the list and recenter password. This did not happen at all with the original knomi

oMTEo, 2023-11-19:
I can also confirm this bug to my Knomi V1. After poweroff and on it 1st shows the Logo+circle then back to Hello-Page.

irgendwer92, 2023-11-23:
Same here on my Knomi V1

Foose83, 2023-11-23:
Same here on my Knomi V2

Foose83, 2023-11-29:
Hello. I entered a shorter ssid and it works correctly now.

oMTEo, 2023-11-29:
Ive tested now some WLAN-configurations with the presence of a team-member of BTT at discord and my results are : Passwords longer than 14 chars wouldnt keep hold after a powercycle of the Knomi. My SSID was only 3 chars at all tests.

I think (and hope) there will come an firmwareupdate soon :)

3dbrooks, 2023-11-29:
This is very exciting, will test with a shorter password. Thanks

MKloberg, 2023-12-01:
@oMTEo Wow, very cool find! My SSID is 19 characters long and the password 15 characters.
I can't change any of that without affecting 33 other devices on my network, so I also hope that this will be fixed soon now that we know what the problem is.

3dbrooks, 2023-12-01:
I changed my Wi-Fi password to 14 characters and my ssid is 6 characters and the issue remained. A second V2 behaves exactly the same. I have exhausted what I can do at my end. 
  I even removed the antenna, Wi-Fi password and put knomi next to my router so it was the only network it saw and even that did not resolve issue.

oMTEo, 2023-12-01:
Damn that thing suckz really, I was sure we´ve found the Bug.

3dbrooks, 2023-12-05:
BTT sent me new firmware and it works!

emintjeee, 2023-12-05:
BTT sent me new firmware and it works!

bigtreetech, 2023-12-06:
Hi. The latest version of firmware should solve the bug of not saving WiFi when password or name are long.
KNOMI2:
https://github.com/bigtreetech/KNOMI/blob/master/KNOMI2/Firmware/knomi2_firmware.bin
KNOMI1:
https://github.com/bigtreetech/KNOMI/blob/master/KNOMI1/Firmware/knomi1_firmware.bin
Follow the steps below to update:
https://github.com/bigtreetech/KNOMI/#firmware-update

closed 2023-12-06 by bigtreetech

Guilouz, 2023-12-08:
@bigtreetech No change, same issue on KNOMI1

milo1987, 2025-03-16:
same problems here! Wifi with 8 letters works, 14 not.