bigtreetech/KNOMI #41: How do I Update Firmware?

BuschyX, 2023-11-18:
There are no .bin files on the firmware page and Visual Studio doesn't show any files that it can open.
I'm obviously missing something

mcbeenb, 2023-11-19:
I was able to generate the firmware following these steps:
1) Download and install VS code
2) Install platformIO extension (View->Extensions)
3) Install git https://git-scm.com/downloads
4) Grab the project form github by either cloning or downloading
5) Inside the project, Open Knomi -> Knomi1 -> firmware -> platformio.ini
6) Click build in the menubar

The bin files will be saved into .\KNOMI\KNOMI1\Firmware\.pio\build\esp32dev

BuschyX, 2023-11-19:
> I was able to generate the firmware following these steps:
>
> 1. Download and install VS code
> 2. Install platformIO extension (View->Extensions)
> 3. Install git https://git-scm.com/downloads
> 4. Grab [...]

Thanks!

mcbeenb, 2023-11-19:
To upload the firmware to a Knomi1, you must go into dfu mode to connect.

Make sure your environment is ready, and you are getting serial timeouts when trying to connect. The next step must be done quickly.

Choose upload in vscode
hold down the reset button on knomi
insert usbc cable
release the reset button

This puts the device in DFU mode, like many other esp32 devices.

You should see the IDE begin to upload.

BuschyX, 2023-11-19:
> To upload the firmware to a Knomi1, you must go into dfu mode to connect.
>
> Make sure your environment is ready, and you are getting serial timeouts when trying to connect. The next step must be done [...]

I did the firmware upload via the web app and I did your way, neither threw any errors or confirmations really. Nothing to say it was already up to date. Just success. And from what I can see there is no way to check firmware on the device from the device. So I am just happy that I did not brick it in the process.

Thanks again, very helpful

mcbeenb, 2023-11-20:
Couple more quick things for anyone else attempting to update firmware, or as in my case, flashing a KNOMI1 to KNOMI2 firmware. 

* There are two different firmware branches on the site currently. If you want to install the KNOMI2 branch, you need to select that branch when downloading. You can select this by either selecting the branch in the upper left of the code view or click the link in the readme.md that will take you to the other tree (https://github.com/bigtreetech/KNOMI/tree/firmware)
* Once you download this branch, you will need to set the env param in platformio.ini to KNOMI1. If you don't do this, you will get cpu mismatch errors when compiling.
* Flashing the KNOMI1 to KNOMI2 gives you the new hello page along with some of the new KNOMI2 features. Obviously there is no touch interface, but a lot of the realtime readouts don't appear to be pack ported either. At least from what I have seen so far. Very cool to see KNOMI1 getting a bit more love.

Hiccup-kzh, 2023-11-20:
Why do I encounter errors when I click on compile firmware
MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package
![Uploading 8ERKC`U$]JNH{FM_LTT[{{4.png…]()

jeffalanmcclain, 2024-09-10:
> Why do I encounter errors when I click on compile firmware MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package ![Uploading 8ERKC`U$]JNH{FM_LTT{{4.png…

I get the same error!  Trying to build a custom Knomi2 with different start up logo from the VORON gif.  Wish there was a way it could just be included as a resource in a directory and pulled in rather than having to convert the new GIF to hex and put in include .c file.