bigtreetech/KNOMI #42: Accelation display for Z shows almost alway full bar

lukicsl, 2023-11-25:
I have installed Knomi 2 on my Voron 2.4.
When running print the x any are showing some "reasonable" acceleration bar, the z bar is nailed to the top the most time, and sometimes s just drops 1-2 bars below max.
[image]

bigtreetech, 2023-12-06:
This is normal because there is gravitational acceleration (9.8 m/s²) in the z-axis direction

lukicsl, 2023-12-06:
Is this a joke or a bug? If not a joke, why not subtract the earth's g-force or at least make it configurable if somebody prints on a space lab or the moon?

jgefaell, 2023-12-07:
Me three. As it is, the Z-axis data is useless.

bigtreetech, 2023-12-08:
> Is this a joke or a bug? If not a joke, why not subtract the earth's g-force or at least make it configurable if somebody prints on a space lab or on the moon?

Got it. will change the logic here and remove the z-axis gravity acceleration

bigtreetech, 2023-12-15:
The latest version of firmware has subtracted 9.8 in the z-axis direction.

closed 2023-12-15 by bigtreetech