bigtreetech/KNOMI #46: No saving and connection

MasterTrandem, 2023-12-01:
Hello,

I have the problem that the Knomi 2 don't save the wifinetwork. I change the ssid length but thats not helping. He only save the Ip from klipper. 

The other problem is that Knomi says the printer is not opartional. When i reboot klipper you can see that Knomi have connection and show the reboot animation, but after thsn nothing. I have a Sonic Pad.

MasterTrandem, 2023-12-01:
Okay i found the problems. 1. The length of the moonrake_ip is the problem. Sonic Pad give every printer a own port. That means for section of the ip adress needs more len than 16. The other problem is the restore of the EEPROM. When it restore the password comes like this �zBPZ�K��� out of the EEPROM.

MasterTrandem, 2023-12-02:
So i thing i fix the problems. For the moonraker IP i change in the config.h String moonraker_ip; // "192.168.255.255" max len = 
from 16 to 22. For the reboot Problem  i change 2 things. First i write in the wifi_setup.cpp direct my SSID and password at .sta_ssid und .sta_pwd. Some was writing when he change the lengt iof the SSID the saves in the EEPROM work, because of this point  i change in the wifi_setup.cpp if (!EEPROM.begin()) from 1024 to 4096. I thing the space was not enouth.

oMTEo, 2023-12-02:
`// "192.168.255.255" max len =
from 16 to 22`
its just a comment and wont change anything

`(!EEPROM.begin()) from 1024 to 4096`
Ive tried this, but also doesnt fix this bug

MasterTrandem, 2023-12-02:
Okay than i don't know.
Sorry yes 😂 i forgot to say that i write the ip also inside the wifi_setup, but on the website i can see more than 16. 
I make this with ChatGPT and he say to change this 16 to 22.
[image]

oMTEo, 2023-12-03:
ChatGPT.... XD, next time ask someone with basic programming skills :D

This :**7125** is Port-Direction and dont need to be written, only IP.

An IP -> **xxx.xxx.xxx.xxx** is 15 characters, so a length of 16 is enough.

MasterTrandem, 2023-12-03:
Yes but this was easier.
No i need the port for the Sonic Pad, because you can use 4 printer simultaneously. Every printer gets his own port. When i only give Knomi the IP he says that the printer is unoperative.

MasterTrandem, 2023-12-03:
Do 

> `// "192.168.255.255" max len = 16` its just a comment and wont change anything
> 
> `(!EEPROM.begin()) from 1024 to 4096` Ive tried this, but also doesnt fix this bug

Do you make a hard reset?

bigtreetech, 2023-12-06:
Hi Please try the latest version of firmware, this should solve the bug of not saving WiFi when password or name are long.
KNOMI2:
https://github.com/bigtreetech/KNOMI/blob/master/KNOMI2/Firmware/knomi2_firmware.bin
KNOMI1:
https://github.com/bigtreetech/KNOMI/blob/master/KNOMI1/Firmware/knomi1_firmware.bin
Follow the steps below to update:
https://github.com/bigtreetech/KNOMI/#firmware-update

Guilouz, 2023-12-08:
@bigtreetech Always the same issue on KNOMI1

claudioguareschi, 2023-12-09:
> Hi Please try the latest version of firmware, this should solve the bug of not saving WiFi when password or name are long. KNOMI2: [...]

@bigtreetech 
The new firmware still has a problem of not saving the web interface data correctly.
The issue is in wifi_setup.cpp in the eeprom write function:

  EEPROM.put<knomi_config_t>(0x00 + sizeof(EEPROM_SIGN), knomi_config);

The size of EEPROM_SIGN (4 bytes) is not returned correctly by sizeof(EEPROM_SIGN).

Change to 
  EEPROM.put<knomi_config_t>(0x00 + **EEPROM_SIGN_SIZE**, knomi_config);
as it was correctly done in the rest of the code.

The older firmware version worked just fine after this change, without the need to convert strings to char[]

Coolbobjoe, 2023-12-17:
Has this been fixed? I tried the stock firmware, no dice, i tried flashing the new firmware nothing, so far ive had 0/4 brand new BTT products have limited or absolutely no functionality. Getting close to the point of sending back the entire shipment!

Jinxted, 2023-12-26:
This has not been fixed in the most current firmware. The issue is we need to be able to enter the port. Not everyone is using a printer that comes stock with vanilla Klipper and Moonraker. People using printers like the K1 Series or the Sonic Pad are stuck needing to have a port set in order to communicate with Moonraker appropriately. Without the ability to tell the Knomi which port to access, we have paperweights. I wish I had not updated my Knomi firmware because the ancient version allowed the port to be set within the ip address. Now I have a Knomi I cannot use.

Jinxted, 2023-12-26:
I take that back - for the v1, the link in the readme and such that points to the "latest" firmware has not been updated to include the actual newest firmware release. I flashed the firmware from https://github.com/bigtreetech/KNOMI/tree/master/KNOMI1/Firmware and it now has a field for the port and is functioning correctly. Update to the newest firmware in the actual repository and do not use any of the current links that download a pre-zipped file with existing older firmware.