bigtreetech/KNOMI #49: Kinomi v2 OV2640 camera interface

Xerusen, 2023-12-14:
Is there any way to use OV2640 interface? kinomi 2 was advertised to be able to use camera and humidity sensor
currently software doesn't take advantage of that

Xerusen, 2023-12-14:
There is also lis2dw12 accelerometer mentioned, but no instruction on how to use any of those features

Coolbobjoe, 2023-12-17:
Following as I also bought this with these interests in mind!

azak1r, 2024-03-30:
would love to use my 3do nozzle cam from the knomi... saves on wire runs

dukes4, 2024-06-23:
Any update please ?

BTopbas, 2024-07-17:
Up

salbang, 2024-07-21:
Up

benbender, 2024-08-01:
@bigtreetech any news?

thejoshuahenry, 2025-03-26:
Apparently this person got the camera working: https://github.com/OmeeeChan/Knomi-V2-camera-function/tree/main

500Foods, 2025-03-26:
Kinda vague on the details tho 😀

thejoshuahenry, 2025-03-26:
Indeed... I assume just flash that firmware, connect a compatible OV2640 camera, then... navigate to the Knomi web address and grab the webcam stream. Just guessing - will need to get a camera module to try.

500Foods, 2025-03-26:
We'll just take it on faith that some random firmware.bin loaded into a device with WiFi and camera hardware is good to go? I mean, sure, I like it, but it just strikes me as funny. But then again, maybe this is the guy who originally designed it, so who knows! Still funny 😆