bigtreetech/KNOMI #50: Potential short when installing knomi v2 according to manual!! - destroying your can board

k4ndr3, 2023-12-20:
Manual flawed at pages 9, 10, 11
Expirienced the consequences 1st hand. Was not paying attention and destroyed my canbus board and the toolboard. 

here is the problem:

[image]

the manual even shows you the consequence

[image]

KoiosLabs, 2024-07-09:
I put some uv solder mask on mine, would recommend similar... electrical tape... kapton... anything

donaldkieu, 2025-04-23:
Just blew up 2 daughter boards because of this antenna