bigtreetech/KNOMI #51: Unable to compile latest firmware

rahsaankcole, 2023-12-24:
[image]

thijstriemstra, 2023-12-30:
Use this pull request to fix it: https://github.com/esphome/AsyncTCP/pull/6

eg:

```
lib_deps    =
              LIS2DW12=https://github.com/stm32duino/LIS2DW12/archive/refs/tags/2.1.0.zip
              Adafruit_SHT4X=https://github.com/adafruit/Adafruit_SHT4X/archive/refs/tags/1.0.3.zip
              lvgl=https://github.com/lvgl/lvgl/archive/refs/tags/v8.3.7.zip
              bblanchon/ArduinoJson@^6.19.4
              TFT_eSPI=https://github.com/Bodmer/TFT_eSPI/archive/refs/tags/v2.5.0.zip
              ; https://github.com/me-no-dev/ESPAsyncWebServer.git
              ; https://github.com/ayushsharma82/AsyncElegantOTA.git
              ottowinter/ESPAsyncWebServer-esphome @ ^3.1.0
              https://github.com/bigtreetech/AsyncElegantOTA.git
              ; Use https://github.com/esphome/AsyncTCP/pull/6 to fix
              ; https://github.com/esphome/issues/issues/5258
              https://github.com/HeMan/AsyncTCP.git#CI-and-fix
```

thijstriemstra, 2024-01-11:
> I had these errors when attempting to compile. I

I also ran into that issue, seems it's broken with 6.5.0 platform so use 6.4.0 instead:

```ini
platform = espressif32@6.4.0
```

BoothyBoothy, 2024-01-11:
> I also ran into that issue, seems it's broken with 6.5.0 platform so use 6.4.0 instead

Thank you, I will give it a try later

BoothyBoothy, 2024-01-11:
Thank you again, that worked to an extent but now get the error you previously commented on in the thrtead. I am really still not quite sure how you implement the suggested PR? Is it simply a matter of copying the lib_deps info to a specific file? I could not see an obvious place to paste? Thanks again Boothy
[image]

thijstriemstra, 2024-01-11:
put PR url in `lib_deps` like I showed in comment above: https://github.com/bigtreetech/KNOMI/issues/51#issuecomment-1872545192

BoothyBoothy, 2024-01-11:
Thank you thank you........I was being really thick there:) Has now complied successfully!

mkryuchkov, 2024-05-26:
> > I had these errors when attempting to compile. I
> 
> I also ran into that issue, seems it's broken with 6.5.0 platform so use 6.4.0 instead:
> 
> ```ini
> platform = espressif32@6.4.0
> ```

You saved my day!