bigtreetech/KNOMI #54: Clarify licensing on the Knomi project.

shishu94, 2024-01-15:
Hello, I would like to submit some changes to the knomi source tree. Or at least fork it.
Some of my changes were well received by the community so I would like to share this work.

Unfortunately, due to my current job, I'm constrained to only submit/fork a project if it's license is explicitly public. The knomi project doesn't include any licensing so it is all rights reserved by default. It would be nice for the community and the product itself to have this project fully open source (probably including the square line studio project as well) as it would allow a larger adoption and use cases from the community. 
I won't  request any specific licensing, but defining a clear one would be nice to understand if the lack of it is just an oversight or the company actually wants to keep ownership (which I respect as well). That would at least clarify the situation and avoid confusion for people like me that would like to tweak the product and share without fear of potential problems.

Thanks in advance.

CBoismenu, 2024-02-15:
It would be realy nice if they changed the license so you can publish your fork of the firmware. It could even get people buy more of the Knomi display.

c4u88sLRM, 2024-02-18:
It would be nice if they even just replied.