bigtreetech/KNOMI #58: Unable to get connect KNOMI or KNOMI 2 to Voron using OctoPrint

TitanColossus, 2024-01-26:
Not sure if this is an issue with the Klipper Port (not sure what this is) or if it is the fact that I am using OctoPrint

Keeps saying Printer is Unoperational Please Check your Printer on the KNOMI 2 and the KNOMI 1 stays on the standby screens.

TheSin-, 2024-04-07:
This is Octoprint, Octoprint has a buffer and doesn't use the virtual SD, moon raker won't set the printing status unless virtual SD is used.  I'm in the same boat and I've been trying to thing or a way to fake it.

TitanColossus, 2024-04-07:
I ended up just reinstalling and using Mainsail and moonraker. Works great now.

amangasas, 2025-01-31:
you can use https://github.com/cp2004/OctoPrint-VirtualPrinterSettings/releases/latest/download/release.zip
if you use moonraker