bigtreetech/KNOMI #62: Knomi 2 V1.0.2 Accel Display Y and Z reversed?

MKloberg, 2024-02-07:
I'm not sure if anybody else has seen this yet. New one, but bought a while back - haven't actually used it until yesterday.
I flashed it with the latest firmware.bin and put it on the printer.
While it completes a layer, I see blips on the x and z but next to nothing on the y.
I don't use z-hop, so there should be next to nothing on the z, unless going up to the next layer.
I see almost nothing on the Y, almost if the two were reversed.
Is anyone else seeing this?

MKloberg, 2024-02-11:
Please ignore my previous issue, I just didn't understand how the Knomi renders these bars.

During printing, it will display three acceleration bars, one for each axis.
Looking at the source code, I learned that the values that are driving these bars are not coming from Klipper. Instead, the sensor that makes this possible is a LIS2DW12 3-axis MEMS accelerometer and mounted on the Knomi PCB itself, sensing how the Knomi moves through 3D space.

If you read this far, you most likely have a bed-slinger style printer, and the print head will only accelerate in the x and z directions while it moves during printing (never in the y direction because that is the axis of the bed). The acceleration bar for the Y axis will therefore never really move and remain mostly at zero.

There should be a configurable option to specify if the printer is a bed-slinger or corexy style printer.
Consequently, for bed-slingers - perhaps there are values on the Moonraker api for each axis that could be displayed to make this somewhat interesting instead of the LIS2DW12 accelerometer values.

austinrdennis, 2024-05-07:
Wow, I agree. This might be the worst implementation of a feature I've ever seen. Agreed, just hook it into moonraker and have it look at instantaneous velocity changes on each axis. This decision is just kinda wild.