bigtreetech/KNOMI #63: Knomi 2 Macro G28 called recursively

rmilyard, 2024-02-10:
When trying to home printer with sensorless-homing,cfg we are getting error: Macro G28 called recursively

What is going on?  I believe I have newest firmware that was posted 2 months ago.

Biffes, 2024-02-27:
Got this problem too. Maybe it is a collision between the sensorless.cfg and the knomi macro. Mainsail console logs about five g28 commands while the error happens.

Specker, 2024-02-29:
I was able to solve this. 
if you have a `[homing_override]` in your `printer.cfg` you will have to change `G28` in there with `G28.1`. 
Similar to mine:
[image]

Biffes, 2024-03-02:
Thanks for the reply, Specker!
I tested your solution and it worked! But i had to change the G28s in the sensorless.cfg too. I found another solution after looking for a fix to the bed mesh calibration for kamp here:
https://github.com/bigtreetech/KNOMI/issues/25

I simply changed the makros in my sensorless.cfg like this:

{% if home_all or 'X' in params %}
    SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=True
    _HOME_X
    SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=False
  {% endif %}
  
  {% if home_all or 'Y' in params %}
    SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=True
    _HOME_Y
    SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=False
  {% endif %}
  
  {% if home_all or 'Z' in params %}
    SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=True
    G90
    G0 X150 Y150
    G28 Z
    G1 Z10
    SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=False

Worked perfectly and i dont have to mess around with modified G28s.

jeffalanmcclain, 2024-09-11:
It is STUPID that this is still an OPEN issue and affecting even pretty base setups.  Thanks for your help figuring this out (I have same issue on bed mesh too).  This is on FlashForge 5M Pro running KlipperMOD.  And I managed to recompile new FW with a new graphic for the start too!
[image]

jeffalanmcclain, 2024-09-11:
Ok, I found a slightly BETTER way to do this.  I don't like having to edit other config files when I add things (like knomi).  And I really don't want things to break if I stop including knomi.cfg with the redefined macros intercepts.  And, it looks like the problem is that [homing_override] is a pretty special macro that does NOT go through the macro rename checks first, it just calls RAW G28 even if you redefine G28.  And it doesn't work to redefine G28 rename old to G28.1 THEN create [homing_override] last.  HOWEVER, Klipper doesn't seem to complain if you REDEFINE [homing_override] again later.  So, I cut and paste the original [homing_override] from my config that runs earlier, and paste it at the bottom of my knomi.cfg and change it THERE to have the hard coded G28.1 calls and it works beautifully (it would be better if defining the routine AFTER the rename G28 to G28.1 would just work with G28, but it don't).  Now, if I ever want to remove knomi, I just remove the [include knomi.cfg] and the original [homing_override] never gets stomped and everything works.

xdased14x, 2024-10-20:
> Ok, I found a slightly BETTER way to do this. I don't like having to edit other config files when I add things (like knomi). And I really don't want things to break if I stop including knomi.cfg with the redefined macros intercepts in the revised [homing_override]. And, it looks like the problem is that [homing_override] is a pretty special macro that does NOT go through the macro rename checks first, it just calls RAW G28 function even if you redefine G28 (it doesn't call your new macro). And it doesn't work to redefine G28 rename old to G28.1 THEN create [homing_override] last (I was hoping it would try to resolve whatever was in place at that point, but it doesn't). HOWEVER, Klipper doesn't seem to complain if you REDEFINE [homing_override] again later (and, no, you can't do rename_existing: in homing_override..Klipper doesn't allow it...sigh). So, I cut and paste the original [homing_override] from my printer.base.cfg that runs / includes in printer.cfg earlier, and paste it at the bottom of my knomi.cfg and change the G28 to G28.1 THERE locally and it works beautifully. It would be better if defining the routine AFTER the rename G28 to G28.1 would just work with G28, but it don't. Now, if I ever want to remove knomi, I just remove the [include knomi.cfg] and the original [homing_override] that is still untouched never gets stomped and everything works.

can you show me the knomi cfg you use trying to fix mine

jeffalanmcclain, 2024-10-20:
well, mine is quite a bit different from anything anyone uses because I have a pretty custom status for LED control and a bunch of other things all broken out...but I can add it here.  Doubt it helps.

`[gcode_macro _KNOMI_STATUS]
variable_homing: False
variable_probing: False
variable_qgling: False
variable_heating_nozzle: False
variable_heating_bed: False
gcode:

[gcode_macro _STATUS_KNOMI]
gcode:
	{% set status = printer['gcode_macro _STATUS_VARIABLES'].printer_status %}
#	{ action_respond_info("Printer status in KNOMI.CFG is: " ~ status) }
	{% if status in ["READY", "PRINTING", "BUSY", "HEAT_SOAKING", "HEAT_SOAKING_END"] %}
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=False
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=False
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=qgling VALUE=False
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=heating_nozzle VALUE=False
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=heating_bed VALUE=False
	{% elif status == "HEATING_EXTRUDER" %}
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=heating_nozzle VALUE=True
	{% elif status == "HEATING_EXTRUDER_END" %}
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=heating_nozzle VALUE=False
	{% elif status == "HOMING" %}
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=True
	{% elif status == "HOMING_END" %}
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=False
	{% elif status == "HEATING_BED" %}
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=heating_bed VALUE=True
	{% elif status == "HEATING_BED_END" %}
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=heating_bed VALUE=False
	{% elif status == "BEDMESH" %}
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=True
	{% elif status == "BEDMESH_END" %}
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=False
	{% elif status == "EVELING" %}
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=qgling VALUE=True
	{% elif status == "EVELING" %}
		SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=qgling VALUE=False
#	{% elif status == "CRITICAL_ERROR" %}
#	{% elif status == "NOZZLE_CLEANING" %}
	{% endif %}
`

selrahc13, 2025-03-08:
> can you show me the knomi cfg you use trying to fix mine

You are a genius. This worked perfectly, thank you.

selrahc13, 2025-03-08:
> Ok, I found a slightly BETTER way to do this. I don't like having to edit other config files when I add things (like knomi). And I really don't want things to break if I stop including knomi.cfg [...]

You are a genius. This worked perfectly. Thank you.

SamppaD78, 2025-05-07:
As to be expected with BTT products,issues that will be never resolved by BTT support. I have exactly the same issue with knomi 2, macro G28 called recursively error. In the process of finding the issue with it updated Klipper and mainboard mcu but still the same issue. As i have a fully working config i dont want to mess with modifying printer .cfg files,so  I was looking for a simple solution as this one below...problem is that i cant find homing_overide in my printer.cfg? I have a voron 2.4r2 with clickyprobe and stock endstops. How to solve this,any help will be appreciated. Thanks.

> You are a genius. This worked perfectly. Thank you.