bigtreetech/KNOMI #70: SCREEN BLACK AFTER UPLOAD GIF

CAIOCDK1, 2024-05-17:
My KNOMI V2 had a black screen after uploading a new GIF, I can't do anything else, not even access the webpage to connect, It also no longer appears as an option on the wifi, nothing happens, I remove the USB and insert it, and only the red light appears in board, not even the Hello message. I can only upload it through VSCODE, but it doesn't seem to have any effect, it completes the process with the SUCCESS message, but the screen remains black.

Rami-Pastrami, 2024-05-18:
Same thing happened with me with 2 separate Knomi v2 boards

electrocally, 2024-05-19:
This also happened to my Knomi v2 yesterday. 

I originally used the web UI at `http://knomi.local` to flash the new gif, following a review & tutorial from [HobbyistNotes on YouTube](https://www.youtube.com/watch?v=EckKEoBZAGw&t=193s). Exact same issues as described above by @CAIOCDK1 with no screen or WIFI network.
I've attempted to use both the Flash Download Tool (same error code "2-sync fail" as @Rami-Pastrami) [documented here](https://bttwiki.com/KNOMI.html#update-firmware), and VSCode using PlatformIO.  Although it has no effect, PlatformIO's output insists that it is able to both write & confirm the hash of the new firmware.

Output snippet:
```
...
Wrote 2939840 bytes (1413934 compressed) at 0x00010000 in 36.2 seconds (effective 650.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
================================================================================================================ [SUCCESS] Took 70.66 seconds ================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
knomiv2        SUCCESS   00:01:10.661
================================================================================================================= 1 succeeded in 00:01:10.661 =================================================================================================================
 *  Terminal will be reused by tasks, press any key to close it.
 ```

electrocally, 2024-05-19:
**I have a fix!**

Posted to the BigTreeTech channel of the Voron Discord & saw a few others had the same issue previously.

The firmware update documentation is for the Knomi **v1**, and hasn't been updated for v2.

At the start of the flashtool screen, select `ESP32-S3`
[image]

From this repository cloned locally, you can flash the default firmware in the master branch. Be sure to select that in the  `KNOMI2/Firmware` folder.

One important difference is that the `knomi2_bootloader.bin` needs the memory address `0x0`. Select your COM port from the dropdown menu.
[image]

Rami-Pastrami, 2024-05-19:
> **I have a fix!**
> 
> Posted to the BigTreeTech channel of the Voron Discord & saw a few others had the same issue previously.
> 
> The firmware update documentation is for the Knomi **v1**, and hasn't been updated for v2.
> 
> At the start of the flashtool screen, select `ESP32-S3` ![flash-tool-board-select](https://private-user-images.githubusercontent.com/15048148/331862800-4b2cc91a-b260-41a0-9f67-4a6ad65093cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxMzM2NTIsIm5iZiI6MTcxNjEzMzM1MiwicGF0aCI6Ii8xNTA0ODE0OC8zMzE4NjI4MDAtNGIyY2M5MWEtYjI2MC00MWEwLTlmNjctNGE2YWQ2NTA5M2NmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE5VDE1NDIzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI4NGNjZmM0YzM5MWJhY2JkMjMzYzQ0MjE4ODMzMzBjNmMyMTc4Y2JiOGE5ZjE1OTAwNjkwODEzZmNjZGM4NjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VysDE2kz6WZFJFWgGkj3cggoP_JuJleWmeUV-48TE20)
> 
> From this repository (cloned locally), you can flash the default firmware in the master branch. Be sure to select that in the `KNOMI2/Firmware` folder.
> 
> One important difference is that the `knomi2_bootloader.bin` needs the memory address `0x0`. Select your COM port from the dropdown menu. ![flash-tool-config](https://private-user-images.githubusercontent.com/15048148/331862802-3a857bb4-0777-4542-8b7f-2b25404dfdc9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxMzM2NTIsIm5iZiI6MTcxNjEzMzM1MiwicGF0aCI6Ii8xNTA0ODE0OC8zMzE4NjI4MDItM2E4NTdiYjQtMDc3Ny00NTQyLThiN2YtMmIyNTQwNGRmZGM5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE5VDE1NDIzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdlMWM4ZTk1NmM3NGU1ODdlYzNmNmRhZTMwM2I3NzI2ZTcxYzYyNWRhYTYxOWEzN2EwZDgwNDM0MjI2MjIxZjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.1R2kd8Lz_EpZE7f9uxMWIHao1nxxOhOZitKQz_KXCig)

This works on bringing back the knomi v2 from death with the stock firmware! Thanks for the heads up!

The one thing remaining is that any firmware I compile fails again with the black screen issue. I thought maybe the gif I was using was at fault, but even when I change 1 C file with one of the original gifs, it still fails like this

electrocally, 2024-05-19:
I was originally just very happy to have my Knomi working again, but I also couldn't get it to run the firmware I compiled.

I'm now leaning towards the conclusion that my board, or perhaps this newer batch of boards that we have, are faulty. The box mine arrived in says "D/C: 2024.2.24" on.

Once mine worked from my comment above, I tried the web UI updater using the same firmware.bin I'd used with the USB flashing tool -- it didn't crash or brick at the time. _but_ when I turned my printer off and on, the toolhead (SB2209 RP2040) wouldn't start & everything smelled like solder.
Seems to be short-circuit protection and the Knomi has somehow shorted out.

My Knomi no longer runs, not even in DFU mode, nor has a red light over USB, but does get **extremely** hot when plugged in. 
I'll be reaching out to BTT's customer service.

CAIOCDK1, 2024-05-20:
> **I have a fix!**
>
> Posted to the BigTreeTech channel of the Voron Discord & saw a few others had the same issue previously.
>
> The firmware update documentation is for the Knomi **v1**, and hasn't [...]

**This solves the problem, my Knomi came back, and I had already done this process, with just one different detail: the instruction number in the bootloader, I was using 0x1000**

closed 2024-05-20 by CAIOCDK1

mkryuchkov, 2024-05-26:
Still can't get something other than black screen with compiled firmware, even stock option (same commit as precompiled one). Any suggestions/solutions?

mkryuchkov, 2024-05-26:
Answer which worked for me https://github.com/bigtreetech/KNOMI/issues/51#issuecomment-1887664967

spigrex, 2024-07-24:
Did anyone ever figure a way around this. I have the same issue. Pre compiled firmware works fine (ever others custome pre compiled firmware worked!) But when I flash anything I compiled/built, I just get the black screen. Tried everything I can find through the issues. I assume it must be a version or something thats out of sync as firmware that others have compiled works on my screen. It's very frustrating and I can see others are having the same issue (which kinda makes me feel a little better about the situation haha)

mkryuchkov, 2024-07-25:
@spigrex look my latest comment

> Answer which worked for me https://github.com/bigtreetech/KNOMI/issues/51#issuecomment-1887664967

TL;DR: use 6.4.0 framework

```
platform = espressif32@6.4.0
```

spigrex, 2024-07-25:
What does this do sorry? Like how do I run this in a terminal? This seems like it fixes an issue with compiling/building errors and has nothing to do with the knomi resulting in a black screen after flashing (sorry if I'm being stupid here!)

spigrex, 2024-07-26:
> @spigrex look my latest comment
> 
> > Answer which worked for me [#51 (comment)](https://github.com/bigtreetech/KNOMI/issues/51#issuecomment-1887664967)
> 
> TL;DR: use 6.4.0 framework
> 
> ```
> platform = espressif32@6.4.0
> ```

Ah you star! This totally solved my issue ...can stop pulling my hair out over this now! ...just changed the env:knomiv2 in platformio.ini and it flashed fine 

Thanks ...feel so silly that it was this simple.