bigtreetech/KNOMI #78: Knomi v2 Can't flash through VS code

borysne, 2024-07-07:
I'm trying to flash Knomi v2 but unfortunately nothing works! The firmware is built in VS code using the platformIO , but when I flash the screen is always black, I tried different firmware, all the firmware that is built through PIO does not work! I tried to flash the firmware built by PIO via USB, and flash_tool did not work! To restore Knomi after firmware failures, I use Flash_tool and flash the Bin file from the official website! I've already tried everything and can't figure out where my mistake is??? please help
I recorded a video, maybe you’ll see something that I didn’t notice!  https://youtu.be/3zP8yZpm3ds

spigrex, 2024-07-22:
I'm also having this issue, I've tried other custom firmware and they work, so I assume it's something I'm doing wrong. I'm not new to vscode/platform io, but this has me totally stumped. 
I've tried using an older version of vscode and platform io just in case that was it, and used an older laptop with a fresh install of older versions of vscode and platform io and the newest versions with no luck. Always black screen. I'd live to know how some people made their custom firmware and what versions they are using.

spigrex, 2024-07-22:
I fixed my issue at least ...seems dumb that I didn't think of it lol. I included the bootloader and partitions bin files when flashing using the esp flash download tool (esp32-c3) I used 0x10000 for all three and it worked 
[image]

borysne, 2024-07-23:
Hello! Is there a video of how you flashed it and it works? can you use OTA to update? access to USB will be closed when you assemble KNOMI into a case

spigrex, 2024-07-23:
Il try and get a video later today. So ota still resulted in the same issue. Black screen. The only way I could flash it was using the tool and the three files.
I had a long play around with it last night, and no matter what I do,  the gifs don't change. I edited the gif files to yellow instead of red, converted them, copied them into the orinigal location. Compiled, flashed, all seemed to work, but all the gifs were the same red color, which makes no sense. So I tried totally different gifs (just a pair of yellow blinking eyes to replace gif_standby) and compiling fails with an error regarding the gif ...dunno if its down to size or something. I gave up after that. 
Im getting very frustrated with it lol.

borysne, 2024-07-23:
I haven’t been able to do anything for 3 weeks now!) This is very strange! It seems to me that we are not reflashing anything at all, but simply restoring the device! because when the device is flashed, it must reset the memory of the wifi settings! but this doesn’t happen to me! the device remembers its ip and login!

borysne, 2024-07-23:
I think when I reflash there should be a white screen as the default!

borysne, 2024-07-23:
I've already tried everything! the only last option that I haven’t done is to try to compile the firmware via Arduino and flash it

borysne, 2024-07-23:
I still don’t understand! You managed to compile the firmware via platformIO! and flash it via flash_tool?

spigrex, 2024-07-23:
Did you try it with all three files included? (Bootloader, partitions and firmware) clearly something isn't quite right still as the gifs just don't seem to change (though the knomi will at least start). If I use any other gifs (same size etc) I just get an error when compiling. Though I'm hoping that's something I can figure out later

borysne, 2024-07-23:
Of course I tried 3 files! I tried my own firmware, I tried other ready-made firmwares! everything doesn't work as soon as I build the firmware via platformIO! Only the original bin from the knomi website works for me and the user's bin file https://github.com/JonathanBourassa/Unofficial-ZeroG-KNOMI works.
and if I zeroG build the firmware myself it doesn’t work again(((

spigrex, 2024-07-26:
OK. Problem solved using information from another ticket ...it was embarrassingly simple hahaha 

In platformio.ini towards the bottom, change platform to the below - 
platform = espressif32@6.4.0

[image]

Hope this works for you!

borysne, 2024-07-26:
does it work for you?
Did you manage to create a custom theme and flash it via OTA?

borysne, 2024-07-26:
I got desperate today and have already started collecting stealthburner! It will be bad if I can’t flash the firmware! no access to usb!)))

spigrex, 2024-07-26:
Yes just tried ota and that works as well. Have tried a custom gif which has worked (just changed the standby to yellow lol)

spigrex, 2024-07-26:
The first reflash I did was using the download tool, but I don't think it will make a difference if you have it up and running on the pre compiled firmware

borysne, 2024-07-26:
ok! I'm taking a risk and will try now)))

spigrex, 2024-07-26:
Let us know how you get on!

borysne, 2024-07-26:
I just managed to assemble the standard firmware via platformIO and flash it via OTA!

borysne, 2024-07-26:
you are just a genius!!!
Thank you!)))

spigrex, 2024-07-26:
Aaahhh yeah, happy for you dude, drove me insane trying to fix it haha.
Can't take all the credit saw another issue that had the awnser, but it didn't actually say what to do with it haha.

borysne, 2024-07-26:
I hope this topic will help other users thanks to you!!!
Now I can try to create custom firmware!)))

borysne, 2024-07-26:
[image]

borysne, 2024-07-26:
Excellent!!! You genius!!!)))

closed 2024-09-11 by borysne