bigtreetech/KNOMI #81: Knomi V2 Won't Connect To WPA2-Only SSID

gailahan, 2024-11-21:
I have a Knomi V2 that will not connect to a SSID if it's configured for WPA2-only.  If I set the SSID up for WPA/WPA2 it will connect.  What's interesting though is when I look at the connection the AP says it's connected via WPA2 using AES encryption.  Why does the SSID need to be set for WPA/WPA2 in order for the Knomi to connect using WPA2?  Has anyone else experienced this?

Bhardwick007, 2024-12-17:
> Why does the SSID need to be set for WPA/WPA2 in order for the Knomi to connect using WPA2? Has anyone else experienced this?

I'm having the same issue with WPA2/WPA3

nomadsgalaxy, 2025-01-30:
Same issue here, i use Unifi, and so I don't have the option to run WPA/WPA2, only WPA2 is minimum spec.

looxonline, 2025-01-30:
@BTTJUSTIN Please put this on the P0 list for the first meeting when everyone is back in the office.

nomadsgalaxy, 2025-02-14:
Any updates on this? I've been trying to fix this on my Livestreams but nothing I've tried is working.

bigtreetech, 2025-02-14:
> Any updates on this? I've been trying to fix this on my Livestreams but nothing I've tried is working.
> 
> The issue appears to be rooted in the inability to join a WPA2 only network

So we ran a heap of testing and here are the results. TLDR, there were a few cases where it could not connect but none of them match your scenario. Are you able to identify *exactly* what connection scheme your router is using?

Knomi Testing Results
Huawei Router TC7102:

OPEN: ✅ Connected normally, Knomi in Open Auth mode.
WPA2 PSK mode: ✅ Connected normally, Knomi in WPA2-PSK mode.
WPA/WPA2 PSK hybrid mode: ✅ Connected normally, Knomi in WPA2-PSK mode.
WPA2 PSK/WPA3 SAE hybrid mode: ✅ Connected normally, Knomi in WPA3-SAE mode.
WPA PSK mode: ❌ Unable to connect.
Mobile Hotspot:

WPA2-Personal: ❌ Unable to connect.
WPA3-Personal: ✅ Connected normally, Knomi in WPA3-SAE mode.

nomadsgalaxy, 2025-02-14:
Attached is a photo of my config, using Unifi routers.

I'll try WPA3 and see if that potentially fixes it

[image]

nomadsgalaxy, 2025-02-14:
I feel incredibly silly, I was entirely unaware that the SSID list was interactable. It's connected now.

I'll be sure to let my audience know the resolution.

[image]

nomadsgalaxy, 2025-02-14:
Wonderful, excited to see it! The UI on the K-Touch has been phenomenally intuitive, so I'm happy to hear there's a project to unify the UI

jangrewe, 2025-08-22:
@bigtreetech any news on a firmware update?

I can confirm that the KNOMI 2 does NOT work with a WPA3/WPA2 PSK hybrid network. If i change my APs down to WPA/WPA2 PSK, it connects (wasted a solid hour in trying to get it to connect until it did that), but when i then changed them back to WPA3/WPA PSK, it gets stuck at the white "Hello" screen again.

No fancy SSID or PSK, identical with WPA3/WPA2 and WPA2/WPA - so it has to be the WPA3/WPA2 PSK hybrid mode. And yes, there's less than 3 meters to the next AP, so the reception is excellent.